jotego/jtbin #155: jtcps15 feedback per version / per game

Saroumane, 2020-12-25:
Hello,
 I remember a few weeks ago I had "Saturday Night Slam Masters" running in attract mode for about an hour without problem.
I can't reproduce that with latest jtcps15 core. So I decided to re-test all versions. 
**TL;DR : I have an 128 MB "2 chips" modules, and I get inconsistent results.**

### jtcps15_20201219.rbf

- 'Cadillacs and Dinosaurs (World 930201).mra'
**Finished all 8 levels of the game without crash.** 
A bit of "tearing" in the upper part of the screesn when dashing after having cleared a wave. 
I don't know if original board does the same.

- 'Muscle Bomber Duo Ultimate Team Battle (World 931206).mra'
black screen

- 'Saturday Night Slam Masters (World 930713).mra'
black screen

- 'The Punisher (World 930422).mra'
black screen or crash after 5min playing

- 'Warriors of Fate (World 921031).mra'
Crash at 1st boss

### jtcps15_20201218.rbf

- 'Cadillacs and Dinosaurs (World 930201).mra'
Crash with illegal instruction after 2min playing

- 'Muscle Bomber Duo Ultimate Team Battle (World 931206).mra'
black screen

- 'Saturday Night Slam Masters (World 930713).mra'
black screen

- 'The Punisher (World 930422).mra'
black screen

- 'Warriors of Fate (World 921031).mra'
Crash at character select

### jtcps15_20201212.rbf

- 'Cadillacs and Dinosaurs (World 930201).mra'
black screen

- 'Muscle Bomber Duo Ultimate Team Battle (World 931206).mra'
black screen

- 'Saturday Night Slam Masters (World 930713).mra'
black screen

- 'The Punisher (World 930422).mra'
crash at boot with "adress error"

- 'Warriors of Fate (World 921031).mra'
Crash mid-level before 1st boss

### jtcps15_20201211.rbf

- 'Cadillacs and Dinosaurs (World 930201).mra'
Crash after 2min playing

- 'Muscle Bomber Duo Ultimate Team Battle (World 931206).mra'
black screen

- 'Saturday Night Slam Masters (World 930713).mra'
black screen

- 'The Punisher (World 930422).mra'
crash at boot with "adress error"

- 'Warriors of Fate (World 921031).mra'
Crash at 1st boss

Saroumane, 2020-12-26:
I also read that the game "Street Fighter Zero (CPS 1.5 QSound Hack)" is a good test.

From MiSTer switched off (for a long time) , 
1. I power on and launch the game "Street Fighter Zero (CPS 1.5 QSound Hack)"
2. I play a few rounds (5min), then let the attract mode run for 70 min : no problem at all
3. Then "cold reboot" (via MiSTer menu), launch NeoGeo core, boot Samurai Showdown 5
4. Then **"cold reboot" (via MiSTer menu)**, launch again "Street Fighter Zero (CPS 1.5 QSound Hack)" in attract mode : illegal instruction in the first demo round
5. I **reboot with a physical button** on the MiSTer and launch again "Street Fighter Zero (CPS 1.5 QSound Hack)" in attract mode : illegal instruction appears almost immediately, before the 1st round
6. I power-off, **wait 10 seconds**, power on, and launch "Street Fighter Zero (CPS 1.5 QSound Hack)" in attract mode : illegal instruction appears a few seconds later, during 1st demo round.
7. I power-off, **wait 70 seconds**, power on, and launch "Street Fighter Zero (CPS 1.5 QSound Hack)" in attract mode : same result than step 6
8. I power-off, **wait 5 min**, power on, and launch "Street Fighter Zero (CPS 1.5 QSound Hack)" in attract mode for 10min : no problems at all.

I don't know what to think.
Does it prove that 
- "cold reboot" is not cold enough ? 
- my SDRAM is faulty ?
- my cooling system is not enough ?
- A mix of these reasons ?

jotego, 2021-01-01:
Thank you for the detailed report. Please try today's update and fill the poll here:
https://www.patreon.com/posts/45683205

closed 2021-01-01 by jotego

Saroumane, 2021-01-02:
You did magic, new core is correcting every problem I listed. (So I don't have to buy to new SDRAM... for now :)