jotego/jtbin #156: ADDRESS ERROR on Cadillacs and Dinosaurs (64 MB SDRAM)

Polo2018, 2020-12-28:
After playing for about 20mn on Cadillacs and Dinosaurs, the game froze but the music continued to play.
This message was on the screen : ADDRESS ERROR (see screen shot). 
My MiSTer has a 64 MB SDRAM board. 
[image]

Polo2018, 2020-12-29:
I played Cadillacs again today. And the game crashed exactly at the same place with exactly the same message (and same memory adresses) :
ADDRESS ERROR 
000D0D79
0001424C

The crash took place about 16 minutes after starting the game. I'm more sure about the time now (I used a chronometer). 

[image]

Polo2018, 2020-12-29:
I played Cadillacs a third time. 

The game crashed after about 7mn of playing. 

It's not the same place as before but it's the same message (and the same memory adresses) :
ADDRESS ERROR
000D0D79
0001424C

[image]

jotego, 2021-01-01:
The 64MB module is also prone to these problems. Not as much as the 128MB module, though.
Please try today's update and fill the poll here:
https://www.patreon.com/posts/45683205

closed 2021-01-01 by jotego

Polo2018, 2021-01-02:
With the latest core update jtcps15_20210102.rbf, for Cadillacs and Dinosaurs and The Punisher games, the issue is now a black screen. Even the warning page from Capcom doesn't appear.