jotego/jtbin #157: Add option for exact 60hz for TV display compatibility

InsertDiskTwo, 2020-12-31:
Add a switch that changes core speed from native to exact 60hz, this will make core 100% compatible and run without any stuttering on all displays, including TVs. Thank you!

jotego, 2021-01-01:
About 33% of users are interested in this.
https://twitter.com/topapate/status/1344596377721065472?s=20

InsertDiskTwo, 2021-01-01:
Considering that a big chunk of the 66% are probably CRT users, 33% is not insignificant :)

jotego, 2021-06-18:
I'd like to do this but I don't have the bandwidth to do it. It adds extra work for each core that cannot be ported from one to the next.
I leave it open in case someone wants to step in.

jotego, 2021-12-22:
I am adding it now to games that have problems for most people, like Konami's Kicker. I don't plan to add it to all cores unless I can get help for it.

closed 2021-12-22 by jotego