jotego/jtbin #160: Black screen on Cadillacs & Dinosaurs and The Punisher

Polo2018, 2021-01-02:
With the latest core update jtcps15_20210102.rbf, for Cadillacs & Dinosaurs and The Punisher games, the issue is now a black screen. Even the warning page from Capcom doesn't appear.

Before, I had the issues described in the bug #156

Polo2018, 2021-01-02:
NB : my SDRAM board is a 64 MB one.

jotego, 2021-01-03:
MiSTer is most likely not finding the MAME zip file.

closed 2021-01-03 by jotego

Polo2018, 2021-01-03:
When I replace this core file jtcps15_20210102.rbf by this core file jtcps15_20201219.rbf (and nothing else is changed), the games Cadillacs & Dinosaurs and The Punisher start without black screen.

jotego, 2021-01-03:
I am sorry to hear that. I think your module is a borderline case. It may work with the next update, or it may work tomorrow without us doing anything. 
I'd recommend contacting your seller. I think they will likely replace it. 
Builders are exploring improvements to the SDRAM module design to make it more reliable and I am helping them as much as I can too.