jotego/jtbin #162: ADDRESS ERROR on Street Fighter Zero (64 MB SDRAM board / CPS 1.5)

Polo2018, 2021-01-02:
After a few seconds of starting Street Fighter Zero (and during the demo), the game crashes. 
This game uses the latest core update : jtcps15_20210102.rbf

A yellow screen appears during the crash and contains this message : ADDRESS ERROR and some values (see screenshot). 

This game perfectly works with an older core vesion : jtcps15_20201219.rbf
[image]
.

Polo2018, 2021-01-03:
This game perfectly works for me with an older core vesion : jtcps15_20201219.rbf

In this case, it's probably not a sdram issue.

PokemonGoDB, 2021-01-03:
I receive the same error with a 128MB ram module. Are you getting it for all CPS1 and CPS1.5 games, or just this one?

Polo2018, 2021-01-03:
This yellow screen with ADDRESS ERROR message and some values (register values) : only on Street Fighter Zero (CPS 1.5)

jotego, 2021-01-16:
I am sorry about these problems, but I cannot do anything else about the SDRAMs. I am helping in the development of new modules by several vendors.

closed 2021-01-16 by jotego