jotego/jtbin #163: Black Screen on Cadillacs & Dinosaurs and The Punisher games NO MAME ZIP ISSUE

Polo2018, 2021-01-03:
With the latest core update jtcps15_20210102.rbf, Cadillacs & Dinosaurs and The Punisher games start with a black screen (even the warning page from Capcom doesn't appear).

BUT when I replace this core file jtcps15_20210102.rbf by an older core file jtcps15_20201219.rbf (and nothing else is changed), Cadillacs & Dinosaurs and The Punisher games start normally (no black screen). 

It's not a mame zip issue. 

Some friends of mine also have this issue.

Polo2018, 2021-01-03:
My SDRAM board is a 64 MB one  - not a 128 MB one.

jotego, 2021-01-03:
I see. 64MB modules are less problematic than the 128MB ones but they aren't perfect either. I can only recommend contacting your module seller. There is not much else I can on my end.

closed 2021-01-16 by jotego