jotego/jtbin #166: sound issues on cps1 latest release

mdd45, 2021-01-16:
On 15012021 cps1 release there cracking noises on games like Final Fight Street Fighter 2 and CE. I havent tested the other games yet. Please check the sound of the barrels breaking in Final Fighr intro and also the sounds of hitting especially on special moves in sf games.

jotego, 2021-01-16:
I have recorded this at 192kHz/24 bit from the optical output. 
https://soundcloud.com/user-14108471/final-fight-20210115
It sounds good to me.

mdd45, 2021-01-16:
Please listen carefully again at 0:19 there is cracking-noise. Maybe you should use other speakers or headset for more detail.

odiaboeeu, 2021-01-16:
If using headphones you can really hear this problem, in sf2 it also happens a few times in the score of points when finishing a fight

MJY71, 2021-01-16:
With the stage music in the Street Fighter II series for example (any of them), every time the kick drum hits, it sound like a tambourine taped to the back of a bass drum. There is ringing/jingling attached to the sound of that one particular instrument that should not be there.

The rest of the audio is absolutely amazing though. Vastly improved sound quality.

jotego, 2021-01-16:
For reference: https://www.youtube.com/watch?v=diceCvElGB0

pacoarcade, 2021-01-16:
I agree with mdd45. That youtube video linked sounds fine.

mdd45, 2021-01-18:
With latest release jtcps1_20210117.rbf and the new available sound options, i think the issue has been fixed. Jotego thank you for the updates and efforts.

closed 2021-01-18 by mdd45