jotego/jtbin #167: sound voice issues on Tora e no Michi

wwark, 2021-01-16:
With the last core jttora_20210115, there is some voice sound issues before to start the first level.

Thanks,

jotego, 2021-01-16:
I noticed that too. But I am not sure how long that has been there as I usually skip the intro sequence. I had a first look a week ago and didn't find anything unusual. I will look into it more.

wwark, 2021-01-16:
It is working great with the previous release (jttora_20200731.rbf)
Thanks,

jotego, 2021-01-16:
Technical notes:

The problem still happens if the old submodules are used:
-jt51 @ 48c3d88
-jt5205 @ 4712ab6

jotego, 2021-01-16:
Fixed. Please update to get the good file. Your hint about the July version was really good. The problem was a typo in a bus width.

closed 2021-01-16 by jotego