jotego/jtbin #170: CPS1 : Knights of the Round - Black pixel presents in the begining of the second stage

wwark, 2021-01-17:
You can see it at the right of Lancelot in this screenshot,
[image]

Or under G of "GO" in this screenshot "thanks @Funkycochise",
[image]

This black pixel moved when you moved Lancelot. Sometimes we have more than one.

I check this YT video using a PCB (visible at the beginning of the video)  and it is doesn't seem present
https://www.youtube.com/watch?t=296

The black pixel is present on those versions 96MHZ and 48MHZ of CPS1 core.

Thanks,

jotego, 2021-01-17:
There is a good chance this is a ROM dump issue. Strider has a similar artifact in one of the main character sprites. I tracked it down and it was just stored like that in the graphics ROM.
We could patch the ROM.

wwark, 2021-01-17:
Thanks for your return.
Did you inform Mame regarding the similar issue for Strider ?

jotego, 2021-01-18:
Not really. They have a GitHub account too and I have entered issues but not this one.

wwark, 2021-01-18:
Yesterday, I tested "Knights of the Round" on Mame and the black sprite are present.
Do you have a screenshot of black sprite on Strider ?
I will report those cases to Mame.

jotego, 2021-01-18:
If you look in the issues here you will find it. Look for Strider and deselect the option "is open" in the search box. 
I think they need to mark the ROM dump as bad.

closed 2021-01-18 by jotego

wwark, 2021-01-18:
Thank you for your time !