jotego/jtbin #172: SDRAM Test Core - test runs with no issue, but core reset causes red bar/failure on most occasions

movisman, 2021-01-20:
Hi Jotego,

I randomly stumbled across some unusual behaviour with both your 48 and 96mhz memory test cores.

If I launch the cores and just let them run, I never get any sort of failure.

However, as the background changes to a lighter green, if I press F12 > reset or the 'User' button on the IO board, more often than not (>50% of the time) this results in a red stripe across the screen and sound of failure.

If I reset the core again while the failure is on screen, the test resets and continues with no issue.

I can replicate this very easily. The only way I can make a red bar appear is by resetting the core when the background is a lighter green. It will not happen if the background is a darker green.

I am unsure if this is an issue exclusive to me, or whether this could potentially be a small bug in the core?

Happy to capture a video if you wish to see more. I know you want to move on from this issue but it'd be interesting to see if you can replicate this.

Would you mind taking a look?

Thanks!

Alex-Catalin, 2021-01-25:
I confirm the same behavior on my side. Without using the reset option from the menu, will allow both tests to run without errors indefinitely. As soon as I hit reset option in the menu during any of the 2 tests, I get a red horizontal bar and a high pitched noise.

jotego, 2021-02-05:
I am sorry this was confusing. Although the reset option is shown on the OSD, it is not intended to be used. I should remove it.

movisman, 2021-02-05:
No worries! Thanks a lot for coming back to me, I appreciate it.

So to confirm, this behaviour doesn't mean there is an issue with the SDRAM then? Is it just a side effect of resetting the core when the screen is a lighter green?

If I run the core like you are supposed to and just leave it, I can currently run 48mhz and 96mhz tests with no issues. In the past I had memory related issues on the earlier 96mhz builds (see https://github.com/jotego/jtcps1/issues/66), but the later builds seemed to be more stable. For transparency, I have also upgraded PSU to 4A over stock 2A, and I have an IO board now which means active cooling - perhaps these upgrades made some difference too, i'm not sure. That's the main reason I raised this though - because I had issues previously I was slightly alarmed by the 'failure' red bar and noise when I reset the core. I thought there was a chance that maybe my SDRAM was slightly problematic, especially because of previous issues. I've been offered a replacement but i'll only go do this if I need to. Was going to wait until CPS 2 was eventually released before deciding - if this is more stressful on the memory it'll be interesting to see how my module handles it.

FYI, the above issue happens on both GUI reset and button reset on IO board. Not sure if you can disable both?

Separate question (if you don't mind) - will you be changing the public CPS 1.5 build back to 96mhz eventually, or leave it as 48mhz?

Thanks a lot as always.

closed 2021-03-05 by jotego