jotego/jtbin #173: jtcps1_20210212.rbf : sound glitches in some games

tAz-07, 2021-02-12:
Hello 

I have some sounds glitch depending on the game. 

For example very audible in "carrrier air wing" when you insert coin.

It is the same with hdmi or analog output and with original-filter option set to on or off.

My Mister is up to date.

ItsAtomico, 2021-02-12:
Other games with coin sound issue:

Carrier Air Wing
Forgotten Worlds
Ghouls N' Ghosts
Pnickies 
Quiz and Dragons 
U.N. Squadron

PepAlacant, 2021-02-12:
Some sound glitches in dynasty wars. They include hanging notes when switching from one track to another in the soundtest.

mikey1964, 2021-02-13:
i think its the same promblem as the carrrier air wing coin sound  play sf2ce go to test sound menu
play code -- OBH

maxxsun, 2021-02-13:
knight of round  when the original filter on

jotego, 2021-02-13:
I can hear those cracks. That wasn't there when I ran the tests. It looks like a last-minute change must have broken it. Let me have a look.

jotego, 2021-02-13:
It's fixed now. I should really stop doing last-minute edits. The file will get uploaded in about an hour.

closed 2021-02-13 by jotego