jotego/jtbin #175: Street Fighter II's blanka melody is wrong.

PepAlacant, 2021-02-14:
Hi Jose. I noticed strange differences in other games, but in blanka's song there is something that lacks in melody, and it sounded in other releases of jtcps1.

https://youtu.be/lZWgo2Ak9sE

maxxsun, 2021-02-14:
turn on the original filter

PepAlacant, 2021-02-14:
> turn on the original filter

Turned on and turned off, same issue.

jotego, 2021-02-14:
It seems like some instruments are one octave higher than they should. Let me look at it.

jotego, 2021-02-15:
The pitch is correct. I will record the original board -I have it- and compare with it.

I have verified fully:
-LFO
-Phase Generator

I haven't fully verified the envelope generator yet.

PepAlacant, 2021-02-23:
https://www.youtube.com/watch?v=kFLJLtlAWcA

Refering to this same sound chip. Double Dragon 2 intro theme has incorrect sounds in the percussions.

the video is not the best quality, but I wanted to find the original PCB.

Here is another video with an emulated game. Previous version of DD2 core sounded right.

https://www.youtube.com/watch?v=t4Wmw7-ycwc

jotego, 2021-02-25:
This version reverts some of the changes made to match NukedYT's implementation, but I keep his LFO architecture. Please try it out.

[jtcps1_20210225.rbf.zip](https://github.com/jotego/jtbin/files/6042742/jtcps1_20210225.rbf.zip)

PepAlacant, 2021-02-25:
still sounds at wrong tone.

PepAlacant, 2021-02-25:
A thing is sure, with your previous implementation ,jtcps1_20210204 the tones were correct both in Blanka's music, Birdie`s Music, Double Dragon II percusions. They were correct for sure. Now it has no vibrato in the main melody and the tones are at a point lower. It happens with the shot of the Willow's weapon too.  

Thanks for your big efforts !

PepAlacant, 2021-02-25:
What piece of music in willow is wrong with this core ? I want to test it.

PepAlacant, 2021-02-25:
I found a thing in the track 09H ... It doesn't have the cymbals on the core. https://vgmrips.net/packs/pack/willow-cp-system#15-dragon-cave-stage-4-2

basically the entire soundtrack lacks cymbals.

PepAlacant, 2021-02-25:
It's curious, because the game starts with Cymbals, but they are getting lost. In the soundtest they are completely absents.

jotego, 2021-03-03:
Tested LFO for waveforms 0-2, AM, PM, all amplitudes. It is correct.
Pending checks:
-Envelope generator
-LFO W=3

jotego, 2021-03-05:
I think this one may be ok.

[jtcps1_20210305.rbf.zip](https://github.com/jotego/jtbin/files/6090824/jtcps1_20210305.rbf.zip)

```
diff --git a/hdl/jt51_noise.v b/hdl/jt51_noise.v
index 4143a6b..e9e44bb 100644
--- a/hdl/jt51_noise.v
+++ b/hdl/jt51_noise.v
@@ -96,7 +96,7 @@ end
 
 // Noise mix
 
-assign mix_sgn = eg!=10'd0 && !out;
+assign mix_sgn =/* eg!=10'd0 ^*/ ~out;
 
 always @(posedge clk, posedge rst) begin
     if( rst ) begin
```

jotego, 2021-03-09:
I think this is fixed now.

closed 2021-03-09 by jotego