jotego/jtbin #182: Trojan (Romstar) (trojanr) - Starting level DIP switches don't work properly

ItsAtomico, 2021-02-24:
Option in the OSD menu doesn't correspond to the actual level you start in. It's also impossible to select level 1 after you change the settings once. 

All versions _except_ "Trojan (US Set 1)" (which just has an in-game level select) have this issue. 
 
According to MAME and Arcade Museum the game had 3 hardware DIP switches on the board that determine your starting level.

MJY71, 2021-04-21:
Confirmed. If the DIP switches are left unmodified, they will start on Stage 1, but when cycling through the DIP switches, these are the stages that all versions except Trojan (US Set 1) actually selects:

6 = Stage 6
5 = Stage 6
4 = Stage 6
3 = Stage 5
2 = Stage 4
1 = Stage 3

Stages 1 and 2 are not able to be selected from that point, because after 3, it cycles back to 6 and repeats those same options.

According to MAME's documentation, `0x00` and `0x20` both select Stage 6 and are not treated as valid options. MAME starts counting the actual switches at `0x40`, which is also Stage 6, but it then starts selecting descending stages at `0x60`, `0x80`, and so on:

```
PORT_DIPNAME( 0xe0, 0xe0, "Starting Level" ) PORT_DIPLOCATION("SWB:6,7,8")
PORT_DIPSETTING(    0xe0, "1" )
PORT_DIPSETTING(    0xc0, "2" )
PORT_DIPSETTING(    0xa0, "3" )
PORT_DIPSETTING(    0x80, "4" )
PORT_DIPSETTING(    0x60, "5" )
PORT_DIPSETTING(    0x40, "6" )
/* 0x00 and 0x20 start at level 6 */
```

Perhaps the counting needs to start at `0x40` in the core (or however it is handled). That way `0xc0` and `0xe0` would become valid options.

paulb-nl, 2021-05-01:
As a workaround the invalid options can be added so it will work with something like this:
`<dip name="Starting Level" bits="5,7" ids="NA,NA,6,5,4,3,2,1"/>`

MJY71, 2021-05-01:
Interesting. Didn't know that could be done like that. Thanks!

jotego, 2021-11-21:
Thank you for all your inputs. It is fixed now in commit b4e0c2d

closed 2021-11-21 by jotego