jotego/jtbin #183: [MIST] low volume on jt cores

squidrpi, 2021-02-24:
Noticed the sound volume on your MIST cores are quite low compared to the majority of other cores on the MIST. For example most MIST arcade cores (not yours) are around -12db in volume whereas yours tend to be around -30db or lower. 

There's no volume control on the MIST.

jotego, 2021-02-25:
This is related to my cores having more audio channels, and particularly using FM chips:

1. A tone from an FM chip has less power for the same signal amplitude than a tone from a square-wave chip (PSG)
2. When many channels are used, you need to accommodate the amplitude to the worst addition case in order to avoid saturation

I recently added an on-chip measurement of the clipping that triggers the LED. I adjusted several cores that had clipping in some scenes, so these cores now sound dimmer in general.

There is not really much I can do. I could add an option to increase volume at the cost of clipping, but I think that an external amplifier is a better solution. I eventually bought an amplifier for my MiST.

Note that if you check JT cores that use PSG chips, such as 1942, you will hear loud sounds. Then again, sometimes the same board may be loud or soft depending on the game (not all CPS games have the same volume, though they use the same board).

closed 2021-02-25 by jotego

squidrpi, 2021-02-25:
Bubble Bobble is particularly quiet.