jotego/jtbin #184: MRA: Some MRA is on MiSTer releases and in _alternatives at the same time

theypsilon, 2021-02-24:
These are:
```
Bubble Bobble (Japan, Ver 0.1).mra 
Contra (US 3).mra 
```

From where should we remove them? From `_alternatives`, or from `releases`?

Also, there is a mismatch between the MRA in `mra/bubl` which is `Bubble Bobble (Japan, Ver 0.1).mra` and `bubl/releases` which is `Bubble Bobble (US, Ver 5.1).mra`

These two folders should contain the same MRA file. Should we have `Bubble Bobble (Japan, Ver 0.1).mra` in `mra/bobl` and `bobl/releases` in one side, and then `Bubble Bobble (US, Ver 5.1).mra` in _alternatives? And for contra, we just remove `Contra (US 3).mra` from `_alternatives`?

jotego, 2021-03-05:
Thank you. Fixed in https://github.com/jotego/jtbin/commit/72180ccf5dfacebe205a089a073c26c350c8f673
I just removed the ones in _alt.

closed 2021-03-05 by jotego

theypsilon, 2021-03-05:
Thanks @jotego !

There is still this mismatch:

On `mra/bubl` we got `Bubble Bobble (Japan, Ver 0.1).mra`
On `mister/bubl/releases` we got `Bubble Bobble (US, Ver 5.1).mra`

Since `mra/bubl` and `mister/bubl/releases` should contain the exact same MRAs, something still has to be changed there. Right now, `Bubble Bobble (US, Ver 5.1).mra` is the only one being distributed by the updater. I can suggest a solution later in PR form.

jotego, 2021-03-05:
Ok. I have made them both the Japan version.

theypsilon, 2021-03-05:
Great