jotego/jtbin #186: MRA: Japan Street Fighter II' Champion Edition alternatives don't work

PigSaint, 2021-03-01:
None of the MRA of the three Japanese versions of Street Fighter II' Champion Edition works. Black screen. These are:

```
Street Fighter II' Champion Edition -Japan 920322-
Street Fighter II' Champion Edition -Japan 920513-
Street Fighter II' Champion Edition -Japan 920803-
```

maxxsun, 2021-03-04:
Black screen usually means that MiSTer cannot find the ROM

PigSaint, 2021-03-04:
I've solved the problem thanks to you. MRA files point to zip "sf2.zip|sf2ceja.zip" (or cejb or cejc), but the right place for the ROM is "sf2ce.zip|sf2ceja.zip". The Japanese versions of Champion Edition are not inside the sf2.zip but sf2ce.zip so adding "ce" to "sf2" is required in this line of code.

This is probably and issue for everyone, so these three MRA needs to be fixed in future releases. Now mine run properly.

jotego, 2021-03-05:
Thank you.
I have fixed the files for everyone now.
https://github.com/jotego/jtbin/commit/12ed976937b359b6badbe6b67293547e654fec21

closed 2021-03-05 by jotego