jotego/jtbin #189: Dynasty Wars audio glitch

Aquas77, 2021-03-05:
At the end of levels, particularly it seems when the screen goes black, it's likely this audio glitch will occur. Happens on most of the levels and at the very end of the game during the cutscenes as well.

Examples: 
https://clips.twitch.tv/SuspiciousRudeDeerChocolateRain-RFKkFXLQKsv5hT0X
https://clips.twitch.tv/TameRenownedSwanOSsloth-KA5wYLzmaVBgQcjX
https://clips.twitch.tv/TameRenownedSwanOSsloth-KA5wYLzmaVBgQcjX
https://clips.twitch.tv/FreezingSpoopyCougarAsianGlow-4gLs6cPS4dt8Ifvx

Still 32mb RAM, latest RBF

jotego, 2021-03-06:
Hi Aquas,
When I changed to using the same LFO circuit as the one reverse engineered, I fixed a problem in Willow, but created a bunch of others. I think they all come down to the LFO random (noise) phase modulation.
I am tracking it here:
https://github.com/jotego/jtdd/issues/14
I will close this issue to avoid duplication. I am working on it today (fourth session on this) and I hope to close on this over the weekend.

jotego, 2021-03-06:
I see now that the Double Dragon bug was much older. I am going to check this one separately.

jotego, 2021-03-07:
It's hard to try anything other than the end of level 1:
-Cannot be heard at the end of level 1 with RBF 200910
-Cannot be heard at the end of level 1 with RBF 210223

jotego, 2021-03-07:
I made an update today, which corrects a couple of minor things I found in the noise generator. It could be related to this issue. I haven't been able to reproduce the problem.

Aquas77, 2021-03-08:
I did a full run and didn't run into the issue! Hopefully it sticks. Great work, thanks JT

closed 2021-04-01 by Aquas77