jotego/jtbin #191: Look at 5x crop scaling

jotego, 2021-03-06:


jotego, 2022-01-17:
Apparently this is not needed with recent versions of MiSTer.

closed 2022-01-17 by jotego

Marat-Tanalin, 2022-01-17:
> Apparently this is not needed with recent versions of MiSTer.

Could you be more specific? Thanks.

atrac17, 2022-01-17:
You can achieve 5x crop with ini presets for arcade cores.

Marat-Tanalin, 2022-01-17:
Thanks. So, if I understand correctly, this 5x crop is now available more universally on MiSTer level instead of core level. That’s great.

The “for arcade cores” part is not quite clear though. Do I understand correctly that the MiSTer-level feature is applicable not to all cores? If so, does the MiSTer-level feature actually make the _entire_ subject core-level feature not needed anymore?

(Fwiw, I’m not a MiSTer user, just have some interest in general for now, so I may be unaware of details that a MiSTer user could be aware of.)

InsertDiskTwo, 2022-01-18:
@thread does adjusting these settings crop the picture at 1080p like other cores? Or users are at the mercy of the TV?

jotego, 2022-01-18:
My understanding is that the MiSTer framework takes care of it now. Please ask MiSTer developers for more details.

paulb-nl, 2022-02-02:
The video_freak module in Sys is responsible for 1080p 5x crop and integer scaling so it needs to be added to MiSTer sys in jtframe. 
https://github.com/MiSTer-devel/Template_MiSTer/blob/master/sys/video_freak.sv

Here it is used in the SNES core:
https://github.com/MiSTer-devel/SNES_MiSTer/blob/163d5007307e9cac7cf2f2316fcb595787639233/SNES.sv#L193-L213
https://github.com/MiSTer-devel/SNES_MiSTer/blob/163d5007307e9cac7cf2f2316fcb595787639233/SNES.sv#L318-L320