jotego/jtbin #192: MIST cps1 captain comando illegal inst.

vastago, 2021-03-06:
Saltan estos mensajes y se detiene el juego (aunque sigue sonando la música):
-ILLEGAL INST.  00000D1C
-ILLEGAL INST.  0001A830
[image]
[image]

Otra cosa que por ejemplo ha sucedido es que al resetear para empezar el juego de nuevo, no se escucha el sonido. O se paraliza el juego directamente.

Arranco el juego desde el fichero ARC.
Al arrancar aparecen como dos lineas blancas intermitentes en vertical:
[image]

Mi configuracion: 
- jtcps1_20210223.rbf cambiado a jtcps1.rbf
-El fichero ARC del directorio jtbin Captain Commando (World 911202).arc (creo que es indiferente que le cambie el nombre por uno con menos caracteres).
-La rom generada con mra
-Conexión a tele mediante scart
-Le he pasado a la MIST los dos programas sdram48 y sdram96 durante unos cinco minutos cada uno, todo verde, parece correcto.
-He puesto como core principal el menú oficial de MIST, para elegir directamente el ARC, (para no arrancar desde otro core previamente cargado, por si las moscas quedaba algo de basura en memoria de la mist). Pero no arregla el asunto.

jotego, 2021-03-07:
Esto es un problema de lectura con la SDRAM. En MiST son muy raros. ¿Es una MiST o una mistica? A veces cambiar de versión del RBF puede ayudar, o algo tan tonto como soplarle a la memoria por si se le ha quedado polvo entre los pines.

closed 2021-03-07 by jotego

vastago, 2021-03-07:
Es una MIST 1.3+, supongo que esta versión ya difiere con el nuevo modelo de MIST (el modelo 1.4). Así que me conformaré con los cores antiguos.
[image]

Hoy he hecho otra prueba de juego y se paraliza con el mensaje "1111 EXCEPTION 00FF89A2"
[image]

De todas formas está bien saber los limites de la máquina de cada uno, esta semana ha sido un continuo de prueba y error constante y creo que definitivamente el core cps1 no se lleva bien con este modelo/lote en concreto (lo mismo los más antiguos puede que funcione alguno, ya los probaré).