jotego/jtbin #204: Strange artifacting in Street Fighter 2

chanunnaki, 2021-03-26:
Doesn't seem to occur in other cores, at least not that I have noticed. I've just seen it in SF2 and SF2:Turbo so far.

Basically, single pixel green vertical lines all across the image. It might just be something to do with my setup. I will do some more testing and see if I spot any more. 

[image]
[image]

chanunnaki, 2021-03-26:
It's happening in all CPS1 games.

chanunnaki, 2021-03-27:
I'm connecting to a Dell 2007FP in 1600x1200 resolution over VGA port with the following video mode:
video_mode=1600,64,192,304,1200,1,3,46,162000

I can confirm it doesn't happen over VGA

jotego, 2021-03-28:
I think this is not related to the core, but to MiSTer scaler logic. So maybe the right place for this issue is [here](https://github.com/MiSTer-devel/Template_MiSTer). They may have it fixed already, I don't know. I update the MiSTer framework every 4-6 months because it tends to be a disruptive process. I will update the framework again in the summer.

closed 2021-03-28 by jotego