jotego/jtbin #207: After jtsf_20201204.rbf Street Fighter (US, set 1) stopped working (black screen)

dfilskov, 2021-04-05:
From jtsf_20210102.rbf and on this core has stopped working.

I have a 64 MB SX v2.2 SDRAM from misterfgpa.co.uk

It runs the JT 96 memtest nicely (and the old memtest from 70 to 150 perfectly and 160 mostly perfect) but fails the JT 48 memtest.

Perhaps there's a way to run this core in 96 MHz as an option to make it work - if that's the issue here?

jotego, 2021-04-06:
It might be possible to speed up that core to 96MHz, but this is not a sustainable solution as some cores need slower clock rates because of internal limitations. It might also be possible to adjust a bit the shift of the SDRAM clock, but again, it is not sustainable as the effective clock shift has a random component that makes it only predictable within a window.

I am quite afraid of touching something that fixes a core for you but breaks it for many other people. But, please try the attached file for "Street Fighter 1", if it boots up, play a bit with it and let me know if there's anything odd.

[jtsf_20210406.rbf.zip](https://github.com/jotego/jtbin/files/6264493/jtsf_20210406.rbf.zip)

Note for me: the file was compiled 5000 ps SDRAM shift.

dfilskov, 2021-04-06:
Thanks so much - I will check :) -  I thought speeding up the SDRAM to 96 Hz as an OSD option could be done in a sort of easy "cheating" way - so that the core still "thought" it was 48 MHz but half the time (every other cycle) the SDRAM was simply "waiting" ... but I'm sure I have misunderstood the issue and know much too little about it :) - I'll check you kindly made new JTSF (and Shinobi) build and let you know! Cheers! :)

dfilskov, 2021-04-06:
Still black screen with jtsf_20210406.rbf unfortunately. jtsf_20201204.rbf was the last build to work properly.

jotego, 2021-05-20:
Please try again with the latest update. You may be lucky this time, but eventually, you will need to replace your SDRAM, I am sorry I cannot do more for you.

closed 2021-05-20 by jotego

dfilskov, 2021-05-20:
Thanks! - I sold the 64 MB RAM module cheaply and all runs nicely with a new 128 MB but I'll ask the new owner :)