jotego/jtbin #241: Street Fighter Zero - CPS Changer alternatives - couple of MRA's to remove

movisman, 2021-07-15:
Hi there,

Hope you are doing well.

Just stumbled across a couple of alternative MRA's I think might be 'surplus to requirements', and thought I should let you know in case a little tidy up is required.

I reinstalled my MiSTer recently, and have the latest MRA-Alternatives_20210709.zip downloaded by the update_all script. So everything is clean, just FYI. Here are the MRA's in question

First one:
_alternatives\_Street Fighter Zero -CPS Changer\Street Fighter Zero -CPS Changer, Japan 951020-.mra

I believe this can go, because the same MRA file now resides in the main releases:
https://github.com/jotego/jtbin/blob/master/mister/cps1/releases/Street%20Fighter%20Zero%20-CPS%20Changer%2C%20Japan%20951020-.mra

So the one in the alternatives folder, isn't really needed now?

Second one:
_alternatives\_Street Fighter Zero -CPS Changer\Street Fighter Zero CPS 1.5 - HBMame.mra

I believe this can go too, because on the 4th June this commit was made to the HBMAME Getter script, which removed sfzch_cps15.zip from being downloaded:
https://github.com/atrac17/MiSTer_MAME_SCRIPTS/commit/89054b6ebc5d8c799f5ae346de58ef064523d4f2#diff-fe5ed5cf1bf440a14ceb1027c341f52e3bfe6d451e7c6c0e94b6c26b3525d66a

sfzch_cps15.zip is required for this alternative MRA, and that file isn't even available on https://archive.org/download/sfz.hack anymore. I assume the hack is no longer available/required.

Would you agree that these can be removed from the next release of the MRA-Alternatives zip?

Let me know if you need any more info.

Thanks as always!

atrac17, 2021-07-15:
I'll see if `sfzch` is duplicated for CPS-1, if so I'll remove the duplicate from the main release. It should fall in the alternates category as CPS-2 is the primary for this game. I understand it's a different version but it's not necessary for more than one mra in main releases. 

Either way it will be fixed with an updated set of CPS-1 mra files. For `sfzch` hack, it will be updated soon with a patched mra not requiring a rom.

movisman, 2021-07-15:
Thanks for the quick reply - and that all sounds good to me.

I did a compare in notepad++ between the "Street Fighter Zero -CPS Changer, Japan 951020-.mra" file in the main releases folder and the alternatives folder and they matched, so I assume one of them needs to go.

As for the hack, that sounds great if it can be patched.

Thanks a lot 👍

atrac17, 2021-07-15:
Sounds good, it's already patched and working and needs to be migrated in with an update some time soon. In the interim it can be found [here](https://github.com/atrac17/Arcade_Offset#current-patched-titles).

movisman, 2021-07-15:
Ah great - I'll check it out shortly! Thanks again.

atrac17, 2021-07-15:
Updated. Old mra files will need to be removed manually for the moment.

closed 2021-07-15 by atrac17