jotego/jtbin #248: CPS2 - eco fighter - missing foreground layer

Spacejockey-GH, 2021-09-13:
En la pantalla 5 faltan todas las columnas verticales que deberian aparecer en primer plano no son dibujadas:

 Core del 20210828: 
[image]

Captura de MAME### 

[image]

Si mal no recuerdo esto se veía bien en versiones previas del core (Antes del fichero beta.zip)

jotego, 2021-09-13:
No, esto fallaba pero se me había olvidado :-(
Lo miro...

Spacejockey-GH, 2021-09-13:
En la primera version del  20210416  se estos pilares se ven bien:

[image]

[image]

Spacejockey-GH, 2021-09-13:
A partir de la version del 20210515 comienza con esta falla.