jotego/jtbin #263: Cotton, graphic glitch, St.2 Moving background

Aquas77, 2022-01-23:
Example video: https://www.twitch.tv/videos/1273148483 (Glitch begins about 10 seconds into the video)

It appears that the moving background of the graveyard, in the latter half of Stage 2, causes a graphic glitch on that background effect which persists until the end of the stage. It is quite easy to pick out once you notice it.

On 32mb xs1.1 sorgelig sdram still, and just ran my update_all.

jotego, 2022-01-28:
I noticed this one when I added Cotton to the beta and tried to fix it back then. As it was the third Cotton GFX glitch in the week, I was already a bit tired and didn't manage to fix it. I'll look at it again. Thanks for the video.

mikey1964, 2022-03-15:
on the title screen it also has that flicker on the clouds maybe its the same bug  looks like the same thing you see on Aquas77 video but in a different part of the game