jotego/jtbin #266: Slow music in Bad Dudes (Dragonninja)

pacoarcade, 2022-02-15:
Since some releases the background music runs slower.

thehughhefner, 2022-03-15:
Please be specific here. Do you have a recording? How is it compared to real hardware?

pacoarcade, 2022-03-15:
To my ears the background music sounds at different tempo in Mister version.

Sure: https://mega.nz/folder/Dw1DFYBZ#OzgQTjpYNRKRcaEM4-aAcw

thehughhefner, 2022-03-15:
Thank you for the videos. Yes, I can hear a difference in the tempo where it is slower in MiSTer vs OG hardware. @jotego

jotego, 2022-03-15:
I think the latest version got a wrong JTOPL (the FM sound) version by mistake. I think it's only a matter of compiling it with the right version. Sorry about the mess. I'll update it.

jotego, 2022-04-17:
This was fixed one or two weeks ago.

closed 2022-04-17 by jotego