jotego/jtbin #268: Only the first two USB controllers work in SiDi

kounch, 2022-02-20:
At least with cores jtpcs1 and jtcps15 (Cadillacs and Dinosaurs, The King of Dragons, etc.), only the first two USB detected controllers work:

- Arcades are configured with tree players option
- Tested all controllers in pairs, with only two players, and they all work ok
- Also tested changing between all the USB ports, with two players, and they work aso ok
- Third player can be controlled without any problem with a keyboard

But, when having tree controllers plugged in, although the firmware sees them, the joystick test pass all ok, etc. the third controller (whichever is detected as third), does not work at all.

kounch, 2022-02-20:
Tried also with several firmware versions between 200614 and 211121, and the issue persists

jotego, 2022-02-23:
I have reviewed the connections and the third and fourth joysticks are connected all the way from the serial interface that links the FPGA and the MCU together to the core logic. I wonder if the firmware is communicating in the expected way. Do you know of a core in SiDi where three joysticks work?

kounch, 2022-02-26:
After doing some more test with 5 controllers at the same time, i see that what the cores use does not match what the firmware is showing on the test screen.

USB controller 1 on test screen -> controller 1 in core
USB controller 2 on test screen -> controller 2 in core
USB _controller 3_ on test screen -> **controller 4** in core
USB **controller 4** on test screen -> _controller 3_ in core

This way, having threee controllers attached will never work with a three player game

kounch, 2022-02-26:
When using the PC-Engine / SuperGrafx, the numbers match, when using 3, 4 or 5 players with Bomberman

USB controller 1 on firmware -> controller 1 in core
USB controller 2 on firmware -> controller 2 in core
USB controller 3 on firmware -> controller 3 in core
USB controller 4 on firmware -> controller 4 in core
USB controller 5 on firmware -> controller 5 in core

kounch, 2022-02-26:
Changed issue text according to latest info

jotego, 2022-04-20:
I have asked for help to the [MiST firmware developers](https://github.com/mist-devel/mist-firmware/issues/90). That's the firmware used in SiDi.

gyurco, 2022-04-20:
It would be helpful to set up ARM debug logging:
https://github.com/mist-devel/mist-board/wiki/Troubleshooting#usb
Then it can be seen what happens when the controllers are enumerated.

gyurco, 2022-04-20:
Actually I could reproduce it in Cadillacs & Dinosaurs input test screen: Controller 3 is ignored, and 3P is Controller 4. I check if it is a firmware or core error.

gyurco, 2022-04-20:
The problem is in JTframe:
https://github.com/jotego/jtframe/blob/master/target/mist/jtframe_mist_base.v#L212
You're using joystick_0,_1,_3,_4, but should be _0,_1,_2,_3

jotego, 2022-04-20:
Blast! It was in front of me all the time. I've fixed it here:
https://github.com/jotego/jtframe/commit/c059a6623a8f914eb8ec7d19b62a3d56a3e2fd45

Thank you, Gyurco!

closed 2022-04-20 by jotego

gyurco, 2022-04-20:
It happens to all of us :)