jotego/jtbin #269: Wrong health bars in Golden axe

pacoarcade, 2022-03-12:
Since some releases Golden axe shows 5 health bars instead of three for normal difficulty. Setting 'Hardest' via DIP switches doesn't change this and there are still 5 health bars instead of one.

jotego, 2022-04-09:
@atrac17 could you check this out, please?

atrac17, 2022-04-10:
Will do.

paulb-nl, 2022-04-14:
Not tested but I think this should be correct:

````xml
<dip name="Difficulty" bits="10,13" values="10,11,14,15,13,7,5,0" ids="Easiest,Easier,Easy,Normal,Hard,Harder,Hardest,Special"/>
````

MJY71, 2022-04-15:
All of the valid options are now present, but are out of order compared against the game's DIP switch settings. When you have that service menu open in the background and cycle through them in the OSD, the only one that matches is 'Normal'.

I'll try to look at them later today if I have the time.

paulb-nl, 2022-04-15:
I think I had the bits reversed. Try this:
````xml
<dip name="Difficulty" bits="10,13" values="5,13,7,15,11,14,10,0" ids="Easiest,Easier,Easy,Normal,Hard,Harder,Hardest,Special"/>
````

atrac17, 2022-04-15:
This is fixed pending other work on System16.

jotego, 2022-04-21:
Thank you for your comments. I see where the problem is and I have fixed it. Except, some values are wrong in MAME too. I have created an issue about it [here](https://github.com/mamedev/mame/issues/9602).
For now, I have fixed it to at least work like MAME. I prefer not to correct the values swapped in MAME and wait until they fix it.

closed 2022-04-21 by jotego