jotego/jtbin #279: Sega System 16A/B Coins/Credits DIP Switches

MJY71, 2022-04-15:
The Coins/Credits DIP switch IDs for the current MRAs contain too many commas. MiSTer treats those commas as DIP switch ID separators so it knows which value to display next in the OSD as they are cycled through. This information cannot always be obtained verbatim from MAME XML data, as MAME uses commas occasionally in DIP switch descriptions for its UI.

Basically, there should always be one less comma than the number of valid options to choose from. In this case, there are 16 possible permutations for each of those DIP switch ranges (each of which is a valid option), so there should only be 15 commas separating those values.

These are the correct values for `DSW1` for all Sega System 16A/B titles:

```xml
        <!-- DSW1 -->
        <dip name="Coin A" bits="0,3" ids="Free Play,2 Coins/3 Credits,4 Coins/5 Credits,5 Coins/6 Credits,4 Coins/3 Credits,6 Coins/4 Credits,2 Coins/3 Credits,4 Coins/1 Credit,3 Coins/1 Credit,2 Coins/1 Credit,1 Coin/6 Credits,1 Coin/5 Credits,1 Coin/4 Credits,1 Coin/3 Credits,1 Coin/2 Credits,1 Coin/1 Credit"/>
        <dip name="Coin B" bits="4,7" ids="Free Play,2 Coins/3 Credits,4 Coins/5 Credits,5 Coins/6 Credits,4 Coins/3 Credits,6 Coins/4 Credits,2 Coins/3 Credits,4 Coins/1 Credit,3 Coins/1 Credit,2 Coins/1 Credit,1 Coin/6 Credits,1 Coin/5 Credits,1 Coin/4 Credits,1 Coin/3 Credits,1 Coin/2 Credits,1 Coin/1 Credit"/>
```

Note: `Free Play` must be selected for both `Coin A` and `Coin B` to be enabled.

These values match the information displayed in real time in the DIP switch service menu screens as they are cycled through in the OSD.

MJY71, 2022-04-18:
My mistake. The above is common to all System 16A/B _bootlegs_.

jotego, 2022-04-20:
Thank you for the detailed explanation. I have fixed it. I will update the repository on Friday.

closed 2022-04-20 by jotego