jotego/jtbin #284: Option separators in <switches>

gyurco, 2022-04-29:
In this patch:
06a6b41e51daca6f69077f672d9f1f44754164ee
How do you make a difference between ';' and ',' as an option separator?
E.g.
Free Play or 1/1,1/1;
should be handled as a whole, but at the end:
6/4,2/3,4/1,3/1,2/1,1/6,1/5,1/4,1/3,1/2,1/1
these are 11 separate options

MJY71, 2022-04-29:
As I outlined in [this](https://github.com/jotego/jtbin/issues/279) issue, MiSTer treats commas as DIP switch permutation separators and therefore cannot be used in the DIP switch descriptions for the OSD, unlike MAME.

What Jotego ended up doing was slightly different than what I suggested. The semicolons are just for display purposes in the OSD. They are not functional with regards to separating one valid DIP switch permutation from the next.

There should be 16 total valid options to cycle through for both Coin A and B.

gyurco, 2022-05-07:
The problem is that semicolon is still used as a separator in MiST. Ok, the MRA tool can be changed to filter out these, but sebdel doesn't seem to maintain it currently.

jotego, 2022-05-09:
The MRA files in d287d35 simply use spaces instead of , or ;

closed 2022-05-09 by jotego