jotego/jtbin #285: MC2P, MC2, NEPTUNO - Options can't exceed 32 - MRA's / Configs being unreachable

ducasp, 2022-05-04:
Hi, I've found this issue with JTKICKER and its variants (Yie Ar, Ping Pong, etc), not sure if that spreads to other bins/mras as well

The code for the microcontroller that parses configurations allows up to 32 configurations to be used, so, you have from 0 to 9 and from A to V to use. W, X, Y and Z will render options after 32 and those options will be impossible to set.

As an example, I can't get Yie Ar Kung Fu to run because service mode is being positioned on option 34, since the MRA is starting with BASE 16 (translates to OG), so, Flip Screen (17th bit, so 16 + 17 = 33), Upright Controls (18th bit) and Service Mode (19th bit) options are impossible to set on the OSD and core will get from the microcontroller something that is a value stuck, that can't ever change.

To work properly on MC2P, MC2, NEPTUNO cores that have options beyond the 32nd bit either need to be reworked to start at a different offset or have some options removed and not go over the 32nd bit of configurations.

jotego, 2022-05-04:
Hi, thank you for the detailed report. MC2/P and Neptuno cores receive little maintenance as there are apparently no patrons backing them. I don't even test them. Unless a certain number of backers appear I cannot really justify spending time on them. 
However, this is a small request and I'll see if I can do something about it.

ducasp, 2022-05-04:
Sure, no problem, I can test MC2P ones, and NEPTUNO should work fine if MC2P does. I fully understand those having low / no priority, thanks anyway for keeping partial support of it and taking some time to look into that.

jotego, 2022-05-09:
I have set the undriven upper 32 bit of the status word in Neptuno to all ones. This should go easier with DIP switches, as the one is normally the off option.
The 8-bit konami cores affected were fixed with this in commit 5081917

closed 2022-05-09 by jotego

ducasp, 2022-05-09:
Thanks for the fix, worked quite nice for Pingpong and Roadfighter and Kicker, but I did not see an Yie Ar Kung Fu updated build, that would be nice (by the way, Yie Ar Kung Fu had its service mode screen upside down, not sure if that is the expected behavior).

jotego, 2022-05-10:
I see the updated build for Yie Ar Kung Fu in the folder, with the other files.

ducasp, 2022-05-18:
Jtyiear has been updated on neptuno bins, but not on mc2p bins. Thanks 🙂