jotego/jtbin #313: 8BitDo Ultimate controller in 2.4GHz disconnects when cores boot

massiveinvisible, 2023-01-22:
When using an 8BitDo Ultimate controller in 2.4GHz using the USB dongle, when I boot cores (tested with AvP, Altered Beast, Alien Syndrome), the controller disconnects just after loading the core. This doesn't happen with a bluetooth controller and isn't happening with any other cores.

jotego, 2023-01-22:
I do not think this is a core-specific issue. The controller communication is handled by the MiSTer, not the core. 
It could be that the ROM loading for these cores creates electrical noise that troubles the Bluetooth communication. The NeoGeo games would create similar noise, I suppose. 
Try moving the USB dongle away from the MiSTer, using a USB hub, for a test.

I'm sorry, but I don't see how else I can help you.

closed 2023-01-22 by jotego

massiveinvisible, 2023-01-22:
Well so far the only cores where I can make this issue occur are yours. I've tried the Neo Geo core and this doesn't happen. Also tried moving the 2.4GHz dongle away from the Mister with a hub and that didn't fix it.
I have found one core of yours where this doesn't occur - Ghouls 'n Ghosts. Is there anything different between that and the others that could be linked to the issue?

massiveinvisible, 2023-01-22:
From testing more cores - it's happening on any that have the "Assembling ROM" progress bar. Doesn't happen with any that don't have that. Still haven't been able to replicate with Neo Geo cores.
Appreciate this may be something happening at a level outside your control but as it's exclusively happening with your cores I thought it was at least worth raising here!

jotego, 2023-01-23:
The _Assembling ROM_ stage happens on all cores, but for most of them, it ends very quickly. The special thing about CPS2 games is that they are large, so the transfer of the ROM files to the SDRAM takes longer. There's a larger chance for electrical interference with the USB dongle in those cares. That's why I thought that the NeoGeo could give you problems too, as those ones are large. But the NeoGeo may be transfering the data in a different way that by mere chance creates less noise in the cases you've tested.

Other than that, all cores are basically identical when it comes to interaction with the gamepad and the USB. Basically, they just get whatever the MiSTer firmware gives to them. 

It was interesting that you raised this. Indeed, I have already seen other cases of the MiSTer being sensitive to electrical noise around, particularly around the SDRAM.

massiveinvisible, 2023-01-23:
Thanks. Just to add to this a few more things I've tried - I tried removing the SDRAM - obviously the games wouldn't boot, but the process for loading the ROMs started and the controller disconnect happened as usual. I also got an OTG hub and tried to use that instead of the USB board - same issue.

massiveinvisible, 2023-01-23:
Related as part of the issue is more of these controller resets - I also just enabled the JT premium cores in update all, ran it then added jtbeta.zip to the MAME folder. When I load Karnov, the image looks off and the menu has a yellow tint. When I reset the game, the controller disconnect happens when I go back to the main menu (this has never happened prior to this - the disconnection only happened when loading a core). After that, loading core gives me no HDMI signal (tested with AvP, Donkey Kong and others) and also the controller disconnect issue happens booting cores where this does not happen before loading Karnov (e.g. Donkey Kong). Powering off and on fixes this, then loading Karnov causes it again.