jotego/jtbin #327: JTsarms refresh rate and service mode working improperly

Missingno255, 2023-03-23:
On the Pocket. JTsarms seems to run at 30FPS. Not sure if that was intended, as the game's refresh rate is 60.00hz. Also, the game resets when turning service mode on, which it isn't supposed to do. It only goes into that sound test and input test screen when the game is manually reset or if the game started with service mode on (like in MAME). You can access several debug features and a level select out of it, by turning service mode on and setting certain dip switches as explained here: [https://tcrf.net/Side_Arms_Hyper_Dyne_(Arcade)](url)

jotego, 2023-03-23:
What makes you think that it runs at 30 fps? Certainly, the core logic operates at the right speed. Could the rest of the Pocket system be doing something with the video that would look like that? Maybe the screen itself? Notice that if the game itself ran at 30fps, everything would be running at half the speed, including bullets and enemies.

Because of how the Pocket interface works, I chose to always reset games when a change in a DIP setting is detected. Otherwise, I have to use a line to add a reset option. Because the number of available lines in the Pocket is so small, I'd rather reset automatically and keep that line to show an option. Also, some users get confused if the options are changed but they don't see the effect, so the reset helps with that.

When games have special features during gameplay if the test service is on, the option text is no longer correct. MAME has an option to display different texts for DIP settings depending upon condition. But that is not possible in the Pocket. At least, not yet.

I have told Analogue about these things in the past but I think they are a bit overwhelmed with the issue and request backlog. So I design the cores around the current Pocket status and assume it won't change.

So, I do not think I can help you much. I see your point about the reset but I think that if I remove it, I will cause more harm than good to most users.

Please let me know what makes you think the video looks like 30 fps when you have a chance.

closed 2023-03-23 by jotego

Missingno255, 2023-03-23:
I compared it with your MiSTer version (closest I could compare to, for obvious reasons), and I do see bullets and player sprites not flickering properly (basically, when the players first spawn in, like the attract mode demo for Stage 7, for example), like they're supposed to when comparing to the MiSTer version. It's not very noticeable, but it caught my eye and figured it might have been a bug with the core.

jotego, 2023-03-24:
Did you compare it using the Dock or using the built-in screen? I have been thinking for a while that the Pocket may have some kind of screen buffer because I have had the impression a few times that the Pocket held the last core image output even when the core is gone. Now, that could be a feature of the LCD screen itself, so I don't know for sure.

I do not have any visibility about the video signal flow once it leaves the core and my experience asking technical details to Analogue has not been positive. So I stopped asking months ago. The only thing I can tell you is that the core outputs the same data in both MiSTer and Pocket.

Missingno255, 2023-03-24:
I used the built in screen, since I don't have a dock.