jotego/jtbin #328: Sorting the Pocket JSON files into folders (`_alternatives` etc)

neil-morrison44, 2023-03-24:
Hey,

With CSP1 & CSP2 growing now we've been thinking about how to sort the JSON files, ideally in a way that allows users to define a subset.

While we could add sorting via the updaters it'd be overwritten by every new core update & would probably mean that core devs get more issues from people running old JSON files etc.

In its simplest form I think the suggestion is to keep the "main" `.json` files at the root as they are now then put extra / alternative ones in a sub-folder similar to [how your mra folder works](https://github.com/jotego/jtbin/tree/master/mra).

If we agree on some naming (could be `_alternatives` like that folder, or whatever works) then we can include a toggle in the updaters to allow users to specify that they want to process the jsons within that folder.

You'd need to move the preset jsons too to match, but I think the content of the json doesn't need to change regardless of how deeply nested it is.

jotego, 2023-03-25:
Yes, I agree that it becomes messy. Could you try a folder layout in the Pocket too and check that it works? The Pocket firmware is a bit buggy and tends to crash for silly things, like empty or unexpected folders.
Please attach a suggestion of a working layout in a zip file to this thread and I'll match it in JTBIN.

jotego, 2023-03-25:
Thanks. You didn't have subfolders in `_alternatives` so _Out Run_ and _Turbo Out Run_ are together. Is it possible to have one more level of subfolders?

neil-morrison44, 2023-03-25:
Yeah that should be fine - there might be a limit of 255 characters for the path length from the root but you can have as much nesting as you want within that I think. 

The updaters will just present a toggle to users to ignore everything under the `_alternatives` level (with it ignored by default to speed things up).

neil-morrison44, 2023-03-25:
I'm just putting a version of Pocket Sync up now that'll (by default) ignore jsons where there's an `_alternatives` **anywhere** in the path so both:
- _alternatives/
    - Outrun/
        - outrun_a.json
        - outrun_b.json
    - Outrun Turbo/
        - outrun_turbo_a.json
        - outrun_rubo_b.json
- Outrun/
    - outrun_main.json
- Outrun Turbo/
   - outrun_turbo_main.json

and

- Outrun/
    - _alternatives/
        - outrun_a.json
        - outrun_b.json
    - outrun_main.json
- Outrun Turbo/
    - _alternatives/
        - outrun_turbo_a.json
        - outrun_turbo_b.json
    - outrun_trubo_main.json

will work fine, or any similar structure (only name that matters is `_alternatives`) so feel free to structure the files however makes sense.
I'll let Matt know and get his & Retrodrivens updaters to do the same

jotego, 2023-03-25:
The underscore in `_alternatives` was a coding needed for MiSTer and it wasn't displayed in the menu. I think that if we use that in the Pocket, it will be shown. We should drop it and just call it `alternatives`.

neil-morrison44, 2023-03-25:
I think the underscore is still a good idea on the Pocket - it means the folder is always in the same place in the file list regardless of what other files are called.

(Also makes it a more uncommon token so we don't accidentally ignore non-alternative files that happen to have alternative in the name)

mattpannella, 2023-03-27:
just chiming in to say keeping the underscore in _alternatives could be useful because it will ensure the folder is always displayed at the top of the list

closed 2023-04-24 by jotego

jotego, 2023-04-26:
@neil-morrison44 could you test the attached zip? If it's fine, I'll move it to JTBIN

[pocket_sorted.zip](https://github.com/jotego/jtbin/files/11336573/pocket_sorted.zip)

neil-morrison44, 2023-04-26:
@jotego 👍🏻 looks good

---

Worth noting that due to the way the installers install the cores (basically just unzipping on top of what's already there) folks'll end up with 2 sets of JSONs for cores already installed.
That'll be on me & @mattpannella to deal with, either via messaging or some sort of automated cleanup - plus CPS2 is the "big one" & having `_alternatives` in there from the start (well, the public release start) should make a big difference

jotego, 2023-04-27:
Is it ok if I change the files in JTBIN tomorrow?

neil-morrison44, 2023-04-27:
@jotego yeah that's fine with me - I'm just deploying a version of Pocket Sync that'll deal with the duplicated files just now so hopefully that'll not be a problem soon (at least for Pocket Sync)