jotego/jtbin #330: X-arcade Tankstick Direct and Xinput not supported

wiggy808, 2023-04-18:
When prompted to register the buttons in the core for the second joystick, nothing registers when moving the directions of the second joystick. 

This occurs for both dinput and xinput modes on the xarcade tankstick. 

Possible solution is to add a core option like what’s in Robotron 2084, where it has the core option “Fire” which lets you choose:
- 4 way
- Move with fire 
- Second joystick

The option “second joystick” in that core allows the second joystick on the xarcade tankstick to be used when in dinput or xinput modes, however no such option exist and current mapping prompts do not allow second joystick inputs on the xarcade tankstick for xinput or dinput modes.

jotego, 2023-04-22:
Could you specify the game or cores for which you need this?

wiggy808, 2023-04-22:
Apologies, I thought this posted directly to the core. It’s for Karate Champ. Some symptoms to help clarify: 1.) The tankstick works fine on direct input or xinput with 2P games like SF2, Final Fight, etc. 2.) The tankstick works fine on direct input or xinput for two joystick games like Robotron. (That core has an option for “Fire” that lets you select “Second Joystick” and map second joystick fire directions for up left right down)3.) When I map buttons for the core, the first joystick maps fine, but the second joystick movements don’t register (while in xinput or dinput).  My hunch is because the tankstick has 1 usb cable for 2 players that a merge / unmerge vid/pid might be an issue / solution, but I’m not sure how to test unmerge or merge for the tankstick for that core. Jonathon