jotego/jtbin #332: Street Fighter Zero 2 Alpha (Japan 960805)

h1deous, 2023-04-28:
As per: https://github.com/jotego/jtcores_mister/issues/9

When you finish a game and enter your name, you’re greeted with the « score ranking screen ». Icons are surrounded by 2 blocks in solid blue color (cf. attached screenshot).
[image]

jotego, 2023-04-28:
I think this have reported before and closed because users found it on real hardware too. The last time it was reported and closed was maybe last week in https://github.com/jotego/jtcores/issues

closed 2023-04-28 by jotego

h1deous, 2023-04-28:
No, it was closed because theypsilon wanted it reported here. It doesn't happen in real hardware. 
See: https://www.youtube.com/watch?v=lDsV1gUsBLo&t=882s

gdlrauro, 2023-12-20:
> I think this has been reported before and closed because users found it on real hardware too. The last time it was reported and closed was maybe last week in https://github.com/jotego/jtcores/issues

Hi, have the CPS2 board of this Game and the blue boxes doesn't exists. If you want i can send you an image direct front my monitor to prove that.
Regards.