jotego/jtbin #340: Fake `jtpatreon` platform_id causing issues

neil-morrison44, 2023-09-02:
Hey,
Looks like the `jtpatreon` platform_id is causing issues with the workflow validation preventing the beta cores from going through (https://github.com/jotego/jtbin/actions/runs/6056813695/job/16437291981)

I think this is a real validation failure since it causes issues with Pocket Sync & likely will with the Inventory API too (and potentially the other updaters).

We're going to look for the existence of the a slot with `name: "JTBETA"` in the data.json to work out which cores are patreon only (and which ones we should put the user's jtbeta key into on setup) - unless there's something better / more stable to look for?

The plan (for Pocket Sync anyway) is to have UI showing the cores as "sponsor only" (or similar) and direct them to the patreon page, plus some handling of the jtbeta.zip file to make sure it goes where it needs to.

jotego, 2023-09-03:
Hi,

Analogue will release a firmware update, hopefully soon, in which all beta cores will be able to share a single `beta.bin` file in the folder `Assets/jtpatreon/common`. The JSON files already work with a beta version of that firmware. But, they also happen to work with the public firmware provided `beta.bin` is copied to the core asset folder, instead of _jtpatreon_'s.

I am glad you can provide an easy way to copy beta.bin to the right location. When the new firmware comes up, I guess it will be safer to just copy it to both locations (jtpatreon and specific core) for a while, until all users update their firmware.

It would be good if the workflow could work with the `jtpatreon` id even now, as this is the way the json files will be anyway.

theYpsilon tag the beta cores as _JTPREMIUM_ in the MiSTer interface. But _sponsor only_, _Patreon only_, _Private beta_, etc. are all perfect.

Thank you so much for your help

neil-morrison44, 2023-09-03:
Is it possible to add a `jtpatreon.json` entry to the platforms?

I’m hesitant to add support for non-existent platforms to the validator since that rule was put there to catch a real scenario where the platform id listed there didn’t match the actual specified platform & that would be lost if it became more permissive.

neil-morrison44, 2023-09-03:
I re-read the Analogue docs and they're pretty permissive about wether a platform needs a json, so I've modified the validator to only show a recommendation.
It'll still error if there's no platform_ids with jsons (since that'll break all the updaters, the inventory, and not allow the user to choose the core in the OpenFPGA menu).

It might be a little while before the inventory & all the updaters support the extra platform_id but the validation should pass next time it's run now

closed 2023-09-03 by neil-morrison44