jotego/jtbin #345: some core files not guilt for mist/sidi

gcopoix, 2023-10-12:
Some of the core files referred in the .mra files can't be found in the mist/sidi folders:
- jtaliens.rbf
- jtkiwi.rbf
- jtoutrun.rbf
- jtroc.rbf
- jtsimson.rbf
- jttmnt.rbf
- jttwin16.rbf
I would appreciate if these cores would be build for the systems above too.
Thank you

jotego, 2023-10-12:
The following cores are in beta phase and are only available via Patreon/GitHub sponsors for MiST and SiDi. The files you see for the MiSTer are not playable without a Patreon key:

jtaliens.rbf
jtsimson.rbf
jttmnt.rbf
jttwin16.rbf

This core is not supported on MiST/SiDi for technical reasons 

jtoutrun.rbf

I think these two should be on JTBIN and somehow are missing. I will add them next week

jtkiwi.rbf
jtroc.rbf

Please consider joining Patreon or GitHub sponsors if you enjoy this work.

jotego, 2023-10-15:
- jtkiwi does not fit in MiST/SiDi
- Added jtroc in commit 59228a4164220d4c8d237f0c4f60d9d53ea71ed9

closed 2023-10-15 by jotego