jotego/jtbin #346: CPS2 Progear Rapid Shot Non-functional

jasoncarlough, 2023-10-27:
There is a button assignment for Rapid Shot in the Controls menu but it is completely non-functional in game.

jotego, 2023-11-02:
Enter the setup mode (F2 on MiSTer or right button+start button) and navigate until you get to this screen. Then enable it.

[image]

closed 2023-11-02 by jotego

jasoncarlough, 2023-11-02:
I can’t seem to get into setup mode. Does it work on Analogue Pocket?

jotego, 2023-11-02:
[image]

jasoncarlough, 2023-11-02:
Ooohhh… sorry about that. I was holding right on the d-pad. Thank you so much for your help and patience.