jotego/jtbin #354: CREDITS WHEN I RUN CPS1.5

valdeaceras, 2023-11-02:
Hi!

    I'm new with the pocket. Maybe it's a stupid question, but when I run cps1 I see a credits that avoid me to play. How can I run the games?
[image]

Thanks in advanced
Regards

valdeaceras, 2023-11-02:
Sorry cps1.5, not cps1

jtmiki, 2023-11-02:
Hi! 
Thank you for joining us!
Please have a look at [this page](https://github.com/jotego/jtbin/wiki/Why-I-only-see-the-on-screen-credits%3F) 
It will help you to solve the problem.
If you can't solve it, please let us know.
Thank you.
JOTEGO

valdeaceras, 2023-11-02:
Hi!!

Thanks for the quick response. Now I'm pathreon, and i've followed the instructions in the page you mentioned (download the jtbeta.zip, unzip and copy in Assets\jtcps15\common and all Assets\*\common folders) but it doesnt work. What more have I do?

Regards,

jtmiki, 2023-11-02:
Hi!

So please read [this](https://github.com/jotego/jtbin/wiki/Analogue-Pocket-Cores) and use one of the updaters on the Installation Guide. 
Hope it'll work this time.
JOTEGO

valdeaceras, 2023-11-02:
Sorry to write again. But I have deleted the beta.bin existing in the jcps15 common folder and then I've used the pocket updater , and the program copied the beta.bin file into the common folder again, But it doesnt work. 
The result of the pocket updater is :

Checking Core: jotego.jtcps15
b1f9fe2 is the most recent release, checking local core...
local core found: b1f9fe2
Looking for Assets
Already installed: wofch.rom
Unable to find wofch.sav in archive
Already installed: dino.rom
Unable to find dino.sav in archive
Already installed: mbombrd.rom
Unable to find mbombrd.sav in archive
Already installed: punisher.rom
Unable to find punisher.sav in archive
Already installed: wof.rom
Unable to find wof.sav in archive
Already installed: slammast.rom
Unable to find slammast.sav in archive
Already installed: mbomberj.rom
Unable to find mbomberj.sav in archive
Already installed: puniserh.rom
Unable to find punisherh.sav in archive
Already installed: dinoj.rom
Unable to find dinoj.sav in archive
Already installed: puniseru.rom
Unable to find punisheru.sav in archive
Already installed: wofr1.rom
Unable to find wofr1.sav in archive
Already installed: wofj.rom
Unable to find wofj.sav in archive
Already installed: slammstu.rom
Unable to find slammastu.sav in archive
Already installed: puniserj.rom
Unable to find punisherj.sav in archive
Already installed: mbombrdj.rom
Unable to find mbombrdj.sav in archive
Already installed: wofa.rom
Unable to find wofa.sav in archive
Already installed: dinou.rom
Unable to find dinou.sav in archive
Already installed: wofu.rom
Unable to find wofu.sav in archive
Already installed: dinoa.rom
Unable to find dinoa.sav in archive
Beta key copied to common directory.
Up to date. Skipping core

Delete all the jcps15 folders and update again with pocket updater?

Regards,

valdeaceras, 2023-11-02:
My pocket updater options were : 
[image]

closed 2023-11-02 by jotego