jotego/jtbin #376: Renegade/Kunio-kun schematic issue

TwistedTom, 2024-02-06:
[Renegade/Kunio-kun schematic](https://github.com/jotego/jtbin/blob/master/sch/kunio.pdf)

page 7 "io"
source of 6502 nmi is unknown
line going to G14A.3 and H8.8 is marked "driver?"

can confirm source is connector J2 pin a19 (marked "CNT_VER")
signal originates from EE18B.6 on the bottom board

jotego, 2024-02-07:
Thanks, @TwistedTom. We noticed it and corrected recently in the source code repository (jtcores) but the PDF here has not been updated yet. 
@etriana85 please check that this is the same correction you made a few weeks back.

etriana85, 2024-02-07:
wip

closed 2024-02-07 by etriana85

jotego, 2024-02-07:
Thanks again @TwistedTom. I'll update the PDF files in this repository on Friday.