jotego/jtbin #378: Latest update brakes scaled output on hdmi on MiSTer for several cores

mapfde, 2024-02-09:
Hi,

the following games no longer work properly on my MiSTer stack (analog i/o, de10-nano, usb board). Analog output at 15KHz is fine, but the scaled output on hdmi has a wrong refresh rate. I have seen 100Hz to 180Hz depending on the video_mode setting. It seems like the scaler gets confused.

Affected cores:
Outrun (see attached image)
Street Fighter
Fast Lane
Roc'n Rope
Side Arms
Double Dragon 2

[image]

jotego, 2024-02-10:
How do you bring that menu up?

mapfde, 2024-02-10:
Push the OSD button, then push the "left" directional button on the d-pad.